synfinatic/teensy-dsc #47: RN-XV not getting configured

BVIS222, 2017-08-11:
Hi Aaron,
I've been trying to get your hardware setup and code running on a breadboarded version of your DSC. It compiles, it reads the encoders, but it is not configuring the RN-XV with the network settings. It is just defaulting to WiFly-EZX-a9 with the default IP address of 192.168.1.1. I've used a printout from eagle of your board to figure out what pins go where but I may be missing something.
I'm doing this on a bread board to get it working first and then it will go into the box with my scope drive system. There is very little room left in the box so building it on your PCB is not really an option.
Any help you can give me would be greatly appreciated.

My email address is bvis222@yaoo.com. This will be easier then using GitHub to communicate ... but that's up to you.

Craig

synfinatic, 2017-08-11:
Have you read the instructions here? https://github.com/synfinatic/teensy-dsc/wiki/Manual-0.5.x

It explains the CLI commands to configure the WiFi

BVIS222, 2017-08-11:
> Have you read the instructions here? https://github.com/synfinatic/teensy-dsc/wiki/Manual-0.5.xIt explains the CLI commands to configure the WiFi—

Thanks for getting back to me so fast...
I've tried that again and it does load the info into the eprom but it never uses that data to change the rn-xv to that ssid or ip address. I'm probably not doing something correctly. Could it be that the RN-XV firmware is causing this....version is 4.41.
Craig

synfinatic, 2017-08-11:
What commands are you using to configure the WiFly?

BVIS222, 2017-08-11:
SET SSID SCOPEDOG => Set the WiFi SSID
SET PASS SCOPEDOG => Set the WiFi passphrase
SET IP 10.0.0.1 => Set the WiFi IP address
SET MASK 255.255.255.0 => Set the WiFi netmask
SET CHAN 4 => Set the WiFi channel [0-12]
SET PORT 4030 => Set the TCP PORT [1-65535]
SET RATE 2 => Set the WiFi rate [0-15]
SET TXP 12 => Set the WiFi TX Power [0-12]
SET AUTH OPEN         => Enable WPA2-PSK or no security
SET MODE AP             => Enable AP or client mode
SET ALT 1440 => set ALT encoder resolution
SET AZ 1440 => set AZ encoder resolution
SAVE
GET ALL returns:
SSID: ScopeDog
PASS: scopedog
IP: 10.0.0.1
MASK: 255.255.255.0
CHAN: 4
PORT: 4030
AUTH: OPEN
MODE: AP
RATE: 2
TXP: 12
ALT/Dec: 1440
AZ/RA: 1440
Then MODE DSC to get back to DSC modeThen I look at WiFi list on computer and all I see is WiFly-EZX-a9Cycle power
Then I look at WiFi list on computer and all I see is WiFly-EZX-a9

BVIS222, 2017-08-11:
I got it workingI need to do and AP SAVE to get it
Thanks for the help.
Will now try and connect via SkySafari

synfinatic, 2017-08-11:
Yep, I was just about to mention AP SAVE. 👍

closed 2017-08-11 by synfinatic